ayushsharma82/WebSerial #2: How do you encode your webpage?

rainG92, 2019-11-25:
I am new to esp8266, just a free time hobby. 
I wanted to create a webserial for a network of devices and i succeeded using the native library provided. I wish to add some customization to the webpage if possible to better control devices on the network.

I have done some researches before i decided to ask.
I believe webpage is stored const uint8_t WEBSERIAL_HTML[] PROGMEM and const uint32_t WEBSERIAL_HTML_SIZE is the arraysize of WEBSERIAL_HTML.

My hypothesis:
1.Gzip the html file
2.Run it through some file to hex converter.
3.Run it through some hex to uint8 converter.

I guess my method is somewhat flaws, i can't seem to make it work.
Mind teach me how you encoded yours? It is interesting to know.
A step by step guide is appreciated. I tried for days to figure out how you encoded yours.
Hope to hear from you, Its fine if this somewhat violate your IP on your works.

ayushsharma82, 2019-11-25:
Hi, You can create your own webpage and convert it into a HEX array with this utility: https://ayushsharma82.github.io/file2raw

closed 2019-11-25 by ayushsharma82

rainG92, 2019-11-25:
Thanks for the wonderful tool.

ShyamBhattacharyya, 2023-05-18:
Error 404: Site is not found on https://ayushsharma82.github.io/file2raw
Can you tell us how to do it?

ShyamBhattacharyya, 2023-05-18:
Thank you for your prompt reply and the link. I really appreciate your hard work.