ayushsharma82/WebSerial #7: How to recreate webserial_webpage.h?

PeeJay, 2020-06-08:
How did you create webserial_webpage.h? Could you add the source in for that?

closed 2020-09-02 by PeeJay

eried, 2021-01-18:
Hi, did you solved the issue?

PeeJay, 2021-01-18:
The author added the source code to https://github.com/ayushsharma82/WebSerial/tree/master/ui but I haven't tried to compile it yet.

eried, 2021-01-19:
I see, but I do not see the moment it gets packaged into the code, so maybe there is another tool required. I can see it is a gz package. I was trying to move the plaintext to the ESP filesystem, but failed.

PeeJay, 2021-01-25:
You need Nodejs v12 to build the package, and if you look in [https://github.com/ayushsharma82/WebSerial/blob/master/ui/finalize.js](finalize.js) part of the build process is to create /dist/webpage.h