ayushsharma82/WebSerial #10: Functions after loop() not working

JOHugo, 2020-07-08:
#include <WebSerial.h> causes the Arduino compiler to give an error with all my functions that are located after the main loop(). The compiler is happy when the functions are defined before the main loop().

E.g. this gives a compile error:
----------------------------------------------
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebSerial.h>

AsyncWebServer server(80);

const char* ssid = "wifi"; // Your WiFi SSID
const char* password = "128bitwepison"; // Your WiFi Password
int count;

void printCount(){  
WebSerial.println(count++);
}

void recvMsg(uint8_t *data, size_t len){
  WebSerial.println("Received Data...");
  String d = "";
  for(int i=0; i < len; i++){
    d += char(data[i]);
  }
  WebSerial.println(d);
}

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }
    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    // WebSerial is accessible at "<IP Address>/webserial" in browser
    WebSerial.begin(&server);
    WebSerial.msgCallback(recvMsg);
    server.begin();
}

void loop() {
  printCount();
  delay(1000);
}

void printCount(){  
  WebSerial.println(count++);
}

betyar, 2020-07-19:
I have the exact same issue. For small sketches this is not a problem, just place all functions at the beginning. For larger sketches with multiple files, however, this is a big problem and renders the library unusable.

AlesSt, 2020-08-13:
HI,
first if you are working in other than Arduino IDE tool then yes this is normal behavior. In proper C you need function declarations before they are used. You cannot eat sandwich before one is made (sandwich = printCount(), eat = use it in any part of your sketch before it is declared/defined, made = declaration/definition)

So your sketch would correctly be:
```
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebSerial.h>

AsyncWebServer server(80);

const char* ssid = "wifi"; // Your WiFi SSID
const char* password = "128bitwepison"; // Your WiFi Password
int count;

void printCount();  //FUNCTION DECLARATION => just announcement to your sketch HEY THIS FUNTION EXISTS - DONT THROW AN ERROR

void recvMsg(uint8_t *data, size_t len){
WebSerial.println("Received Data...");
String d = "";
for(int i=0; i < len; i++){
d += char(data[i]);
}
WebSerial.println(d);
}

void setup() {
Serial.begin(115200);
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
if (WiFi.waitForConnectResult() != WL_CONNECTED) {
Serial.printf("WiFi Failed!\n");
return;
}
Serial.print("IP Address: ");
Serial.println(WiFi.localIP());
// WebSerial is accessible at "/webserial" in browser
WebSerial.begin(&server);
WebSerial.msgCallback(recvMsg);
server.begin();
}

void loop() {
printCount();
delay(1000);
}

void printCount(){   //FUNCTION DEFINITION => anything you want this function to do
WebSerial.println(count++);
}
```

What you had in your sketch was 2x function declaration :) 

What I like to do is make a file: declarations.h and I include it in a sketch before first function is declared/defined with `#include "declarations.h"` => needs to be in the same folder as your sketch file
and in that file I put all the declarations of ALL functions in the sketch

Hope it helps :)

ayushsharma82, 2021-04-05:
Solution provided by @AlesSt. 

Closing the issue.

closed 2021-04-05 by ayushsharma82