ayushsharma82/WebSerial #17: Lost messages

plistiak, 2021-01-12:
Hi.
  I found that when I sent more messages via print in short time, some of them are lost. Is there any limitation about sending messages? Here is demo:

```
void loop() {
  for (byte i = 0; i<10; i++) {
    WebSerial.print("test ");
    WebSerial.print(i);
    WebSerial.println();
  }
  delay(5000);
}
```

It should write ten lines, but only 3 is written, as you can see below
[image]
[image]

omar117, 2021-01-27:
Primero que nada es fantástico el proyecto. Mientras hacia pruebas sin necesidad de usar un delay encontré que llegando a un punto casi al instante se bloque los datos, por lo que probé el problema no es por parte del autor sino que es en la librería ESPAsyncWebServer-master en el archivo AsyncWebSocket.h y AsyncWebSocket.cpp. Al parecer al transmitir demasiado rápido se llena una cola que es la que se encarga de guardar los datos y enviarlos, como el programa o la ejecución es muy rápida no le da tiempo de vaciar la cola y se bloquea. Encontré una solución que la verdad no se si es la mejor pero funciona.
Solución:  abrir el archivo AsyncWebSocket.h y modificar la línea 30 #define WS_MAX_QUEUED_MESSAGES 8 (línea original ) a #define WS_MAX_QUEUED_MESSAGES 32 (línea modificada).
Abrir el archivo AsyncWebSocket.cpp y agregar  _messageQueue.free(); en la línea 563.
[image]
Otra cosa que note es que la trasmisión de datos es mas rápida si se usa WebSerial.print((char)Cualquier_Dato) como quien dice transmitiendo solo char, por ejemplo 
WebSerial.print((char)Serial.read()) y no es igual la transferencia si se usa WebSerial.println por que directamente los pasa a String; . Finalmente recomiendo usar el navegador chrome de Android de alguna marea va mas rápido que en el navegador de escritorio,

ayushsharma82, 2021-04-05:
Solution provided by @omar117 .

The limitation is of AsyncWebServer.
Closing this issue.

closed 2021-04-05 by ayushsharma82