ayushsharma82/WebSerial #25: Checking if there is room for more serial characters to avoid error too many messages qued

StefanL38, 2021-04-16:
Hi,

is there a way to check if the WebSerial.println  or WebSerialPro.println  still has room for buffering more characters?

best regards Stefan

ayushsharma82, 2021-04-16:
@StefanL38 This is directly linked to AsyncWebServer's WebSocket buffer. If you can increase it then it will increase the buffer of webserial as well.

pancanet, 2021-10-09:
@StefanL38 you can try to change WS_MAX_QUEUED_MESSAGES value inside AsyncWebSocket.h,  the default value from library is 8 for esp8266 and 32 for esp32 and max value is 256. also, dont push data with webserial.print or .println too often, the library need a time to remove the buffer. 
[image]

ayushsharma82, 2021-11-17:
Solution provided by @pancanet. Closing.

closed 2021-11-17 by ayushsharma82