ayushsharma82/WebSerial #32: Send message via URL

galtamur, 2021-06-09:
Hello, I would like to send a message to the ESP32 by composing a URL, like <ip_address>/webserial/message=<my_message>
Is it possible anyhow?
Thank you,
GA

ayushsharma82, 2021-06-10:
@galtamur  You can use DDNS and port forwarding to make your ESP publically available.

ayushsharma82, 2021-06-10:
Oh sorry, I didn't read your message carefully the first time.

No, it's not possible to send a message via a query in URL.

rkuo2000, 2021-07-02:
> Hello, I would like to send a message to the ESP32 by composing a URL, like <ip_address>/webserial/message=<my_message>
> Is it possible anyhow?
> Thank you,
> GA

You can <ip_address>/get?message="your_message"

Python sample codes: 
[sendHTTPget.py](https://github.com/rkuo2000/homebot/blob/master/sendHTTPget.py)
[sendHTTPpost.py](https://github.com/rkuo2000/homebot/blob/master/sendHTTPpost.py)

closed 2021-08-12 by ayushsharma82