ayushsharma82/WebSerial #37: WebSerial with PubSubClient for MQTT

nadi1971, 2021-10-01:
I'm trying to build a sketch for NodeMCU using PubSubClient for MQTT and WebSerial libraries.
I'm using the following libraries:
#include <ESP8266WiFi.h> 
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
#include <WebSerial.h>
#include <PubSubClient.h>

When I'm enabling WebSerial library, I'm receiving the following error:
D:\Arduino\ESP8266_Async_Demo\ESP8266_Async_Demo.ino: In function 'void setup()': 
ESP8266_Async_Demo:36:22: error: 'callback' was not declared in this scope; did you mean 'calloc'?
   36 |   client.setCallback(callback);
      |                      ^~~~~~~~
      |                      calloc
Multiple libraries were found for "Hash.h"
 Used: C:\Users\home\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\Hash
 Not used: D:\Arduino\libraries\AsyncElegantOTA
exit status 1
'callback' was not declared in this scope; did you mean 'calloc'?

Could you please check how to make webserial library working with PubSubClient library?

nadi1971, 2021-10-01:
[ESP8266_Async_Demo.txt](https://github.com/ayushsharma82/WebSerial/files/7267670/ESP8266_Async_Demo.txt)

ayushsharma82, 2021-10-01:
The culprit is how the webpage was compiled previously.

I'll merge the new script in a couple of days to produce a better `webpage.h` file.

ayushsharma82, 2021-10-03:
Should be resolved with v1.3.0 release, Let me know the outcome.

nadi1971, 2021-10-03:
I've just checked and looks like it working.
I'll try to compile the completed sketch and let you know if I'm still having problems.
Thanks for your effort!

closed 2021-11-17 by ayushsharma82