ayushsharma82/WebSerial #38: Sending Special Characters

belveder79, 2021-10-06:
Hi, thx for this great work. 

I purchased the Pro version and I wonder if there is any option to send special characters like ^C or so over the textbox. I'm using the webserial on an ESP32 to directly interface a PC over a serial port, giving me a login to the PC. Login and password prompt works as expected, so do all commands that terminate. If I do a simple ping X.X.X.X , it obviously does never terminate automatically, so I wonder if I could send ^C eg. over the textbox. 

Is this possible and if yes, how? Can this be integrated somehow if it is not there already?
thx

dannyz2, 2021-11-17:
I see you havn't gotten a response as yet, you could work around the Ping problem, by using a switch with Ping. 
Use "Ping IP Address -n 10" to send 10 requests.

belveder79, 2021-11-17:
thx - yeah, limiting the number makes total sense in this case.

The major question I had was if sending special characters would work somehow... if that was possible, the tool was way more versatile...

belveder79, 2021-11-17:
well, I'm connecting to a PC over the webserial of an ESP32, in order to execute some debugging commands if necessary. Therefore it would be good to be able to send some special characters as well, as e.g. with the ping command to terminate it... that's just one example...

ayushsharma82, 2021-11-17:
@belveder79 Did some quick research, seems like it's not possible to type special characters in HTML webpages. You'll have to build some custom logic on top of WebSerial to interpret incoming messages and apply special characters.

closed 2022-12-19 by ayushsharma82