ayushsharma82/WebSerial #49: Cannot compile anymore even demo code:

gfaman, 2022-06-09:
Hi,
from some weeks already, I cannot compile anymore my own software which was working well 1 month ago.
Even the simple demo version here cannot be compile anymore. Seems some issues with some dependent libraries.
Always got undefined reference:
c:/users/gfama/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\gfama\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/AsyncWebSocket.cpp:1273: **undefined reference to `SHA1Update'**
c:/users/gfama/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\gfama\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/AsyncWebSocket.cpp:1274: undefined reference to `SHA1Final'
c:/users/gfama/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\ESPAsyncWebServer-master\WebAuthentication.cpp.o:(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/gfama/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: libraries\ESPAsyncWebServer-master\WebAuthentication.cpp.o: in function `getMD5(unsigned char*, unsigned short, char*)':
C:\Users\gfama\OneDrive\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'

Any idea ? I tried to downgrade board ESP32, all dependent libraries, cannot find the reason.

Thanks,

Laurent

ayushsharma82, 2022-06-16:
Hi @gfaman ,
Please provide the version of your ESP32 Arduino Core, I'll check it out.

gfaman, 2022-06-30:
Hi, I'm in Arduino 1.8.19
Thanks,

italocjs, 2022-12-19:
Hello, has anyone found the issue?  I'm having the same problem here on platformio.

My libdeps is:
```
lib_deps = 
	ayushsharma82/WebSerial@1.3.0
	me-no-dev/ESP Async WebServer@1.2.3
	me-no-dev/AsyncTCP@1.1.1

```	
and errors:
```
Linking .pio\build\esp32dev\firmware.elf
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x18): undefined reference to `SHA1Init'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x1c): undefined reference to `SHA1Update'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x20): undefined reference to `SHA1Final'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(AsyncWebSocket.cpp.o): in function `AsyncWebSocketResponse::AsyncWebSocketResponse(String const&, AsyncWebSocket*)':
D:\OneDrive - simova.com.br\06 Projetos SIMOVA\SIMOVA_xBTSAT\xWEBSERIAL/.pio/libdeps/esp32dev/ESP Async WebServer/src/AsyncWebSocket.cpp:1269: undefined reference to `SHA1Init'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:\OneDrive - simova.com.br\06 Projetos SIMOVA\SIMOVA_xBTSAT\xWEBSERIAL/.pio/libdeps/esp32dev/ESP Async WebServer/src/AsyncWebSocket.cpp:1270: undefined reference to `SHA1Update'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:\OneDrive - simova.com.br\06 Projetos SIMOVA\SIMOVA_xBTSAT\xWEBSERIAL/.pio/libdeps/esp32dev/ESP Async WebServer/src/AsyncWebSocket.cpp:1271: undefined reference to `SHA1Final'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/italo/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib6a6\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
D:\OneDrive - simova.com.br\06 Projetos SIMOVA\SIMOVA_xBTSAT\xWEBSERIAL/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

mytistrauss, 2023-01-31:
You need to get the latest library for _(NOTICE THE SPELLING!!!)_ ESPAsyncWebSrv. Then change all #include statements from "ESPAsyncWebServer.h" to "ESPAsyncWebSrv.h"

closed 2024-06-12 by github-actions[bot]