ayushsharma82/WebSerial #50: problem using WiFiManager together with WebSerial

MAWMN, 2022-06-18:
Basic Infos
Hello,

I have a problem in using WiFiManager at the same time as WebSerial in a Program, because the two libraries below are NOT compatible with each other..
used for WiFiManager : #include <ESP8266WebServer.h>
used for WebSerial : #include <ESPAsyncWebServer.h>
is there any way to be able to use both, or the functionality of both ?

Hardware
Esp8266

Hardware: Wemos-D1-mini

Core Version: Core 6.0.2·Home 3.4.2

Description
Problem description
see above

Settings in IDE
PlatformIO
Module: Wemos D1

Additional libraries:

Sketch
n.a.

Debug Messages
incompattible libraries

kickshawprogrammer, 2022-06-18:
Issue #42 suggests using AsyncWifiManger instead of WifiManager. Since I'll be running into the same problem, I'll be looking into that solution as well but right now, I've got another issue with WebSerial.

MAWMN, 2022-06-19:
Thanks for the suggestion, I will have a look at it. Was not aware of this library..
Can you tell me which one you are using ?
I Found two .
   https://github.com/lbussy/AsyncWiFiManager
   https://github.com/alanswx/ESPAsyncWiFiManager

kickshawprogrammer, 2022-06-19:
I am not quite to that point yet but from an initial screening, I'd probably go with lbussy's since it forks straight from tzapu's directly. My only concern, initially, is that it hasn't been touched in 2-years. Is that because the project is abandoned or is it just that stable?

Regards,

MAWMN, 2022-06-19:
@kickshawprogrammer

at the moment I am using : https://github.com/alanswx/ESPAsyncWiFiManager
Actually working quite well in combination with WebSerial..
However I have seen a problem that the available networks in my case "OnDemandAP" is not being revoved from my list of WiFi networks in my Phone after being connected to the correct selected network, when I again try to connect to this AP that is not possible ??? 
So why still in my list???
The only moment it is disapearing is when I am uploading a new software version to the device???
Another problem I have seen that in certain occasions it is giving me a WDT reset, while I am only committing to the EEPROM ??
Message...
Soft WDT reset
>>>stack>>> etc...

`
  // RELAY_PIN = AAN
  digitalWrite(RELAY_PIN, HIGH);
  RELAY_IsON = true;

  // Increment Router Resets counter in EEPROM
	EEPROM.write(3, (byte) (EEPROM.read(3) + 1));
  EEPROM.commit();
    
  MonTxt = lastDisplayedDate + " " + lastDisplayedTime;
  MonTxt = MonTxt + " - Router is RESTARTING.....\n";
  MonitorInfo(MonTxt);  // display in Serial port & WebSerial screen

  waitTime = millis();
  while (millis() < waitTime + RELAYS_WAIT_FOR_ON_TIME * SECOND) {
    delay(100);
  }

**** Here the watchdogs triggers ****

  // RELAY_PIN = UIT
  digitalWrite(RELAY_PIN, LOW);
  RELAY_IsON = false;
`

kickshawprogrammer, 2022-06-20:
I can’t help with that first problem. First off, that’s not the library I suggested. But even at that, I have no experience with either library. Furthermore, I’m not even having a conflict between wifimanager and webserial so right now I have no need for either. 

As for your second issue, you’re tripping the WDT in your delay(100) loop. 

The WDT (I believe there’s two) is set to go off after like 1/2 seconds and like 3-seconds if memory recalls. 

In either case, it happens when the core code doesn’t get a chance to do its system stuffs. Time consuming while loops, such as yours will cause it. That’s normal. 

Your loop() function should be able to run a complete cycle in like 100ms or less. After completion of your loop() function, the core gets its chance to do its work, before starting over.  

Since your loop() is taking so long, the system never gets its chance and so it cries foul and throws the WDT. 

Your fix is really simple in this case. Simply replace the line delay(100); with yield(); instead. Yield() gives the system a chance to do its thing. 

The conditions for the other WDT are a little more complicated. Suffice it to say, if you’re still getting WDT’s after changing to yield(), you really need to rethink your loop() function. 

I hope this helps,

MAWMN, 2022-06-20:
Hello,
   the Watchdog problem has been solved, the function was called directly from the WiFiSerial input (Probably interrupts).. Solved by placing it in, and handled by, the main loop by means of a Flag.. Problem solved ;-)

The other problem still persists, still working on it..

closed 2022-06-22 by MAWMN