ayushsharma82/WebSerial #53: ESP32 Memory Leak

darrenlock, 2022-11-09:
I have noticed that each call to WebSerial.println leaks about 20 of Heap. Once the browser connects it then leaks about 40. When the browser disconnects about 900 is lost. See test programme which builds on the example:

`/*
  WebSerial Demo
  ------
  This example code works for both ESP8266 & ESP32 Microcontrollers
  WebSerial is accessible at your ESP's <IPAddress>/webserial URL.

  Author: Ayush Sharma
  Checkout WebSerial Pro: https://webserial.pro
*/
#include <Arduino.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#elif defined(ESP32)
#include <WiFi.h>
#include <AsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>  // https://github.com/alanswx/ESPAsyncWiFiManager

#include <WebSerial.h>

AsyncWebServer server(80);

const char* ssid = ""; // Your WiFi SSID
const char* password = ""; // Your WiFi Password

/* Message callback of WebSerial */
void recvMsg(uint8_t* data, size_t len) {
    WebSerial.println("Received Data...");
    Serial.printf("Received Data... [%d] ", ESP.getFreeHeap());
    String d = "";
    for (int i = 0; i < len; i++) {
        d += char(data[i]);
    }
    WebSerial.println(d);
    Serial.printf(" %s [%d]\n", d.c_str(), ESP.getFreeHeap());
}

void setup() {
    Serial.begin(115200);
    Serial.println("Starting Wifi...");
    DNSServer dns;
    AsyncWiFiManager wifiManager(&server, &dns);
    if (!wifiManager.autoConnect())
    {
        Serial.println("wifiManager.autoConnect() failed to connect - restarting ESP");
        delay(3000);
        ESP.restart();
        delay(5000);
    }
    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
    // WebSerial is accessible at "<IP Address>/webserial" in browser
    WebSerial.begin(&server);
    /* Attach Message Callback */
    WebSerial.msgCallback(recvMsg);
    server.begin();
    Serial.printf("Setup complete [%d]\n", ESP.getFreeHeap());
}

void loop() {

    static uint32_t    _AtStart = ESP.getFreeHeap();
    static unsigned int _lastSend;
    static uint32_t    _After = 0;
    static uint32_t    _Before = 0;    
    static uint32_t    _SinceStart = 0;

    if (millis() - _lastSend > 5000)
    {
        _Before = ESP.getFreeHeap();
        Serial.printf("Last WebSerial Call [%d] Lost Since [%d]\n", _After, _After - _Before);
        Serial.printf("Before WebS [%d] ", _Before);
        WebSerial.println(ESP.getFreeHeap());
        _After = ESP.getFreeHeap();
        Serial.printf("After WebS [%d] Lost [%d] \n", _After, _Before - _After);
        Serial.printf("At Start [%d] Lost Since Start [%d] Lost this iteration [%d]\n", _AtStart, _AtStart - _After, (_AtStart - _After) - _SinceStart);
        _SinceStart = _AtStart - _After;
        _lastSend = millis();
    }
}`

ayushsharma82, 2022-12-19:
Hi @darrenlock ,

Were you able to narrow it down to what's causing this memory leak?

Looking forward to fixing it.

darrenlock, 2022-12-20:
Hi @ayushsharma82 ,

unfortunately I have not had chance to investigate any further. Maybe one of my projects over the holiday :)

jksemple, 2023-02-02:
I didn't see this behaviour when testing my pull request. Not sure if my changes have eliminated a memory leak but if so, I don't know why.

closed 2024-06-12 by github-actions[bot]