ayushsharma82/WebSerial #65: Timing Issue - Need delay between multiple WebSerial.println commands

JMWilton, 2023-05-26:
When executing the following code:
WebSerial.println(WiFi.localIP());
 //delay(500);
 WebSerial.println("joe was here");
 // delay(500);
 WebSerial.println("this is now");

The output is:
[image]

When executing the following code:
WebSerial.println(WiFi.localIP());
delay(500);
WebSerial.println("joe was here");
delay(500);
WebSerial.println("this is now");

The output is:
[image]

It appears that subsequent WebSerial.println commands overwrite the previous WebSerial.println command if there is an insufficient delay.  It would be expected that no delay should be necessary.

JMWilton, 2023-06-08:
I now suspect that the html receive event code might be reentered before it finishes its work.

asjdf, 2023-06-09:
I thought this issue might be help

https://github.com/asjdf/WebSerialLite/issues/4#issuecomment-1342696623

closed 2024-06-11 by github-actions[bot]