ayushsharma82/WebSerial #79: [v2] Add Arrow UP / DOWN in the free version to recall past command from the local storage history

mathieucarbou, 2024-06-02:
Why adding this feature in the free version ?

Because IMO this a really basic feature. For example in an OSS project, to build a remote weight scale for beehives, in partnership with a big M2M company (https://beelance.carbou.me), a web console is used to communicate with the model (AT commands). There are a few other projects that have use cases with a command terminal and communication. These projects have to remain free because they are community programs serving / helping people (in this case beekeepers).

The issue with not having this feature in the free version is that for such project, this is a real blocker. 

The pro version cannot be used, and if the free version does not have it, then another lib will be used (so no promotion of this project).

So I am creating this issue as a feature request in order to log this use case and requirement and see with you if this is possible to get a WebSerial free version for OSS projects with a command history.

mathieucarbou, 2024-06-02:
FYI here is how I implemented it in a fork: https://github.com/mathieucarbou/WebSerialLite/blob/main/frontend/index.html#L213-L361

I will add it to the pro version (and I will also add you to the private repo so that you can grab the changes).

But this issue is more about porting it to the OSS version tool....

ayushsharma82, 2024-06-03:
I'll pull your feature request to OSS version too in a few days.

mathieucarbou, 2024-06-03:
> I'll pull your feature request to OSS version too in a few days.

Thanks!
I have it implemented in the pro version already ;-) Works like a charm!

ayushsharma82, 2024-07-06:
Released with v2.0.4

closed 2024-07-06 by ayushsharma82