ayushsharma82/WebSerial #90: Can't compile - error: no matching function for call to 'AsyncWebServerRequest::beginResponse

PsyonicOne, 2024-07-01:
Using your latest library installed via the PIO library installer.
(from my platformio.ini)
`ayushsharma82/WebSerial@^2.0.3`
`mathieucarbou/Async TCP@^3.1.4`
`mathieucarbou/ESP Async WebServer@2.10.4`

Using PIO with VS Code on Linux

Gives the following error:
.pio/libdeps/esp32-3248S035C/WebSerial/src/WebSerial.cpp:62:119: error: no matching function for call to 'AsyncWebServerRequest::beginResponse(int, const char [10], const uint8_t [20891], unsigned int)'

ayushsharma82, 2024-07-01:
@PsyonicOne Docs are yet to be updated, it will work fine once you update ESPAsyncWebserver to v3.0.5

PsyonicOne, 2024-07-01:
@ayushsharma82 That worked awesome!
Also... that was by far the fastest response to a bug report I've ever had! Thanks heaps mate!

closed 2024-07-01 by PsyonicOne

mathieucarbou, 2024-07-01:
@PsyonicOne : please use:

```
  mathieucarbou/Async TCP @ 3.1.4
  mathieucarbou/ESP Async WebServer @ 3.0.5
```