ayushsharma82/WebSerial #97: function onMessage dont works

tobiasfaust, 2024-08-13:
Hi,
I tried your example Logging.ino as my first test with webserial. But it seems, the funktion onMessage dont works because i dont get the messages as real serial output. Output as webserial works. 
Do you imagine whats wrong? If web connection fails, i need the serial output as real serial output twice ;)

`WebSerial.onMessage([](const String& msg) { Serial.println(msg); });`

tobiasfaust, 2024-08-17:
Hi @ayushsharma82
it would be fine if you could take a look into this?

ayushsharma82, 2024-09-04:
@tobiasfaust I tested logging.ino example on my end and it still works. Message sent from WebSerial is begin printed in `Serial`. Please check your setup.

tobiasfaust, 2024-09-04:
Hi @ayushsharma82 ,
thanks for your reply. 
For a real testcase close to your test 
1. i cloned your repo, 
2. correct a spelling mistake in platformio.ini
  mathieucarbou/ESP Async WebServer@^3.1.1 -> mathieucarbou/ESPAsyncWebServer@^3.1.1
3. compile for esp8266
4. upload, show serial monitor and webserial

and unfortunally, no change. perfect loggin in webserial, the last serial log shows me the first ip-Adress and no more

whats wrong? I havent further ideas...

[image]

ayushsharma82, 2024-09-05:
Okay, so I tested it on ESP32. There must be an issue with ESP8266 part of code. I’ll check that when I’ll get some time in my hands.

@mathieucarbou You wrote this callback, have any idea why it’s not working?

mathieucarbou, 2024-09-06:
@ayushsharma82 : it was fixed in https://github.com/ayushsharma82/WebSerial/pull/92 and released in 2.0.5. I think @tobiasfaust is not using the latest version. I am testing it with `HighPerf.ino` and it works fine.

[image]
[image]

mathieucarbou, 2024-09-06:
For 8266: I saw some issues in the code. HighPerf mode does not compile. I am preparing a PR.

mathieucarbou, 2024-09-06:
ESP8266 works. Tested on D1 mini
[image]
[image]

tobiasfaust, 2024-09-08:
thank you, it works well now with ESP8266

PR #98 (2024-09-07): Project CI update and ESP8266 fixes
- Closes #97
closed 2024-10-14 by github-actions[bot]