ayushsharma82/WebSerial #109: Official code not showing initial prints

hserras, 2024-12-11:
Hello everyone,

Really sorry if this post wont be well formatted, written or even on the right place, but it's the first time I open a post here. If something is wrong, I'm really sorry.
I started using the WebSerial library, currently version 2.0.8, in order to print some stuff there. But the main issue for now is with the "callbacks". The first callbacks are not being printed in the WebSerial UI.

The code I'm using is the official one, and from what I see, everything is fine, but something is wrong.

The callback code from the official site and it's precisely the same than mine:

**Official code**

```
// Attach a callback function to handle incoming messages
WebSerial.onMessage([](uint8_t *data, size_t len) {
  Serial.printf("Received %u bytes from WebSerial: ", len);
  Serial.write(data, len);
  Serial.println();
  WebSerial.println("Received Data...");
  String d = "";
  for(size_t i=0; i < len; i++){
    d += char(data[i]);
  }
  WebSerial.println(d);
});
```

If you see the serial prints that are printed, you see that I clicked 9 times, from 1 to 9:

```
14:37:06.433 -> Received 2 bytes from WebSerial: 1
14:37:07.729 -> Received 2 bytes from WebSerial: 2
14:37:08.309 -> Received 2 bytes from WebSerial: 3
14:37:09.056 -> Received 2 bytes from WebSerial: 4
14:37:10.087 -> Received 2 bytes from WebSerial: 5
14:37:10.680 -> Received 2 bytes from WebSerial: 6
14:37:11.075 -> Received 2 bytes from WebSerial: 7
14:37:11.613 -> Received 2 bytes from WebSerial: 8
14:37:12.097 -> Received 2 bytes from WebSerial: 9
```

[image]

If I see the webserial, which should have barely the same information, the first two entries somehow seem to remain on some sort of "cache", because they are printed later on:

[image]

So, the order of this was:
In Webserial, I clicked 1 and nothing was printed;
In Webserial, I clicked 2 and nothing was printed;
In Webserial, I clicked 3 and it printed 1;
In Webserial, I clicked 4 and it printed 2;

And so on and so on.
Is there an issue with the library itself?
Can someone help out on this, since this is a pretty weird behavior?
I believe my code is fine, since it's precisely the official one

mathieucarbou, 2024-12-11:
I've contributed in the past a high perf mode activated by: `-D WSL_HIGH_PERF=1`.
It will remove any buffering and send the data directly into the websocket queue.
the downside is that it is only compatible with API printing full lines like printf, println, write(buffer), etc.
do not use it to send characters in a loop.

so looking at your code, it should work better.

hserras, 2024-12-11:
Well, at the end, it's not my code, it's the official one :-D
And on this code, there's not a loop, so that shouldn't be a problem.
It's quite weird that with their OOB code, this happens. And like explained before, after the first two entries, everything works normal, everything is printed in the webSerial, but with delayed entries.
Any other suggestion?

mathieucarbou, 2024-12-11:
yes, because the original implementation has a buffering mode and delay. 
It bothered me also so that's why I contributed this mode which bypass the buffering and does like the old webserial was doing. 

Try with `-D WSL_HIGH_PERF=1` ;-)

This contribution, plus some others like key up / key down to recall the commands, come from [MycilaWebSerial](https://github.com/mathieucarbou/MycilaWebSerial) project.

hserras, 2024-12-11:
ok, great. I'll try that, once I can.

Last dummy question, since I never used this, and at the end, there is no example with this crystal clear.
This piece of code/flags should be used where in Arduino IDE? Right at the beginning of the code, before the setup?

hserras, 2024-12-11:
Mmm, after reading a bit, it seems this needs to be added in platform.txt and not in the *.ino file
Can I add this to whatever row in that file, or it needs to be on a specific location of the file?

mathieucarbou, 2024-12-11:
> Mmm, after reading a bit, it seems this needs to be added in platform.txt and not in the *.ino file Can I add this to whatever row in that file, or it needs to be on a specific location of the file?

I don't know about Arduino IDE. I am using Vscode and platformIO / pioarduino, like any other devs. 
Arduino IDE is a toy to play with Arduino, not a real development environment  ;-)

hserras, 2024-12-12:
Hey Mathieucarbou
Well, I completely agree with you. Arduino IDE is a toy in comparison with VS + PlatformIO
I worked some years ago with PIO and my microcontrollers, but left it also some years ago, and I started again a few weeks ago, but I was playing around again with Arduino IDE. But reached the conclusion that yes, PIO is the way to go.
To resume, yesterday, I installed again PIO. And I can tell you that I've tried the flag you mentioned and it worked beautifully.
Thanks a lot for your help.

closed 2025-01-17 by github-actions[bot]