ayushsharma82/WebSerial #112: Consecutive whitespace characters not displaying on web

noisymime, 2025-02-11:
I'm trying to layout some text blocks using multiple spaces and/or tab characters, but these are not correctly displayed on the WebSerial side due to the way HTML treats whitespaces. 

Eg Using the following in a sketch:
```
WebSerial.println("||");
WebSerial.println("| |");
WebSerial.println("|  |");
WebSerial.println("|   |");
```

Results in:

```
||

| |

| |

| |
```

Rather than the expected:

```
||

| |

|  |

|   |
```

I'd suggest that the frontend should use a `<pre>` tag rather than just a `<p>` around the displayed messages as that will result in the correct formatting.

noisymime, 2025-02-11:
Sorry, also meant to mention that the class needs to be changed to `whitespace-pre-wrap`

Eg

From:
`<p class="whitespace-pre-line text-wrap break-words">|   |</p>`

To:
`<pre class="whitespace-pre-wrap text-wrap break-words">|   |</pre>`

This results in whitespace displaying correctly.

noisymime, 2025-03-13:
Any interest in this fix?

ayushsharma82, 2025-03-21:
Hi @noisymime , Thanks for spotting the issue. I'll implement that fix.

closed 2025-03-21 by ayushsharma82

ayushsharma82, 2025-03-21:
Tried the fix and it causes weird artifacts with existing code, therefore I'm reverting your suggestion.

[image]

commit 96bc09f4 (2025-03-21): fix: consecutive whitespace characters not displaying on web - closes #112
commit a5eb74e3 (2025-03-21): Revert "fix: consecutive whitespace characters not displaying on web - closes #112"

This reverts commit 96bc09f4131a72be2b5a42c965c9aaabd4e5c375.